lagadic/vision_visp #51: howto run with kinect one (V2) as camera stream ?

flobotics, 2016-05-19:
hi,
i want to use kinect one (V2) camera stream and camera_info for visp_auto_tracker, but dont know right howto do it.  

- do i need to insert a node which starts the kinect2, or can i simply use the available streams from a already started kinect2 ? there are kinect2/hd/image_color  and kinect2/hd/camera_info topics to which i could subscribe , but how?   remapping, from inside which node ?

- what do i need to insert into the launch file then ?

if some more infos needed i will provide them

fspindle, 2016-05-19:
Try the following

````
  <!-- Launch visp_auto_tracker node -->
  <node pkg="visp_auto_tracker" type="visp_auto_tracker" name="visp_auto_tracker" output="screen">
    <param name="model_path" value="$(find visp_auto_tracker)/models" />
    <param name="model_name" value="pattern" />
    <param name="debug_display" value="True" />
	      
    <remap from="/visp_auto_tracker/camera_info" to="/kinect2/hd/camera_info"/>
    <remap from="/visp_auto_tracker/image_raw" to="/kinect2/hd/image_mono"/>
  </node>
````

flobotics, 2016-05-19:
thanks, that works.

but, 

- it detects the paper-size big QR code only if i get around 40 cm infront of the camera ?

- the overlay appears and then it resizes bigger and bigger, until its around as big as the QR code, and then it disappears, then it appears again, gets bigger and so on and on , how does it get more stable ?

thanks

fspindle, 2016-05-19:
Sounds not clear to me. Could you send me a video

flobotics, 2016-05-19:
https://www.youtube.com/watch?v=5xYyCqz5Ruc  
the resizing-thing is around 1:13 

why does it not recognize the qr-code correct?

fspindle, 2016-05-20:
1/ The QR code should be large enough in the image to get a valid detection. 
2/ The complete target (QRcode + black square) is not well printed. It should be in the middle of the paper with enough white border arround the external black. With your printing, one side of the black border is cutted. Black region should be square nor rectangle as yours.

flobotics, 2016-05-20:
oh, my printer did not correctly printed it in the middle. i printed it new and here is the result
https://www.youtube.com/watch?v=2WJsd6y7dTo

- could the qr-code be smaller? what is the smallest size it can detect ?

- my pc is running at 80% so it will miss some pictures, but when the qr-code is not flickering, why does it not get detected sometimes, even infront of the camera ?

thanks for help

fspindle, 2016-05-20:
1/ It will be worse if the QRcode is smaller. I notice that detection works better if the target is parallel to the image plane. If you want to detect the target when you are far, the only way I see is to print a larger target (just apply a scale to qrcode + back square)

2/ Your framerate is to low. That's why once a detection is valid, the tracking initialised from the detection fails after few images when you apply a displacement to the target. Find a way to acquire at least at 20/30 Hz. If not possible, you could try to increase the range in models/pattern.xml, but not sure it will be enough:

````
    <range>
        <tracking>15</tracking>
    </range>
````

flobotics, 2016-05-20:
the qr-code even needs to be bigger? oh, thats not working for me, perhaps i must test the model-based tracking, because in the online videos they track a tea-pack, which is a little smaller than my qr-code. i would like to track a cube of 2x2 cm

fspindle, 2017-02-02:
Closed since the initial issue seems fixed

closed 2017-02-02 by fspindle