lagadic/vision_visp #52: Using live camera stream as Input data for visp_tracker

TobiasCy, 2016-05-20:
Greetings,

I try to run the visp_tracker with my own camera and I fail. The problem is, that the GUI is not starting.
visp_auto_tracker is working with my camera.
 No i try to exchange the bag stream with a live camera stream in the visp_tracker
I used remap in the tutorial.launch with the following lines:

 
  <node pkg="visp_tracker" type="tracker" name="tracker_mbt">
    <param name="camera_prefix" value="/wide_left/camera" />

    <remap from="/wide_left/camera/image_rect" to="/camera/acA2000_50gc/image_raw" />
    <remap from="/wide_left/camera/camera_info" to="/camera/acA2000_50gc/camera_info" />

    <param name="tracker_type" value="mbt+klt" />
  </node>

When I start the launch file the processes are starting but no GUI apprears:

process[rosout-1]: started with pid [14521]
started core service [/rosout]
process[tracker_mbt-2]: started with pid [14539]
process[tracker_mbt_client-3]: started with pid [14540]
process[tracker_mbt_viewer-4]: started with pid [14541]
process[base_link_to_camera-5]: started with pid [14542]
process[tf_localization-6]: started with pid [14550]

Does anyone know what could be the problem? I changed nothing else than the launch file. The camera is also running under the new topic, since i can echo it. Is it maybe a problem with the calibration?

Kind regards
Lyon

fspindle, 2016-05-20:
The tracker is waiting for a rectified image.
As I understand your camera is publishing /camera/acA2000_50gc/image_raw and /camera/acA2000_50gc/camera_info. To generate a rectified image you should use `image_proc`

The launch should be similar to something like:
````
  <!-- Launch the image_proc node to publish /camera/acA2000_50gc/image_rect -->
  <node pkg="image_proc" type="image_proc" name="image_proc">
    <env name="ROS_NAMESPACE" value="/camera/acA2000_50gc"/>
  </node>

  <!-- Launch the tracking node -->
  <node pkg="visp_tracker" type="tracker" name="tracker_mbt" output="screen">
    <param name="camera_prefix" value="/camera/acA2000_50gc" />
    <param name="tracker_type" value="mbt+klt" />
  </node>

  <node pkg="visp_tracker" type="visp_tracker_client" name="tracker_mbt_client" output="screen">
    <param name="model_path" value="file:///home/user/mbt-models" />
    <param name="model_name" value="teabox" />
    <param name="tracker_type" value="mbt+klt" />
  </node>

  <!-- Launch the viewer (GUI) -->
  <node pkg="visp_tracker" type="visp_tracker_viewer" name="tracker_mbt_viewer"/>
 
````

Here the tracker is supposed to track a model named teabox that is located in `/home/user/mbt-models/teabox` and that contains:

````
$ ls /home/user/mbt-models/teabox
teabox.cao  or teabox.wrl
teabox.init teabox.xml
teabox.ppm (optional)
````

TobiasCy, 2016-06-03:
Thanks a lot, works create!

closed 2016-06-03 by TobiasCy