lagadic/vision_visp #54: visp_auto_tracker cannot open model file when I run with  external camera

wanlong0827, 2016-06-20:
Hi,
I'm trying to use visp_auto_tracker but I'm stuck on this issue:
It has worked when roslaunch visp_auto_tracker tracklive_usb.launch by my computer's camera.
But when I run with external camera of MAV ,I encountered some problems.
I modify tracklive_usb.launch：
<launch>	
  
  <node pkg="visp_auto_tracker" type="visp_auto_tracker" name="visp_auto_tracker" output="screen">
    <param name="model_path" value="$(find visp_auto_tracker)/models" />
    <param name="model_name" value="pattern" />
    <param name="debug_display" value="True" />
	      
    <remap from="/visp_auto_tracker/camera_info" to="/dji_sdk/camera_info"/>
    <remap from="/visp_auto_tracker/image_raw" to="/dji_sdk/image_raw"/>
  </node>

  
  <node pkg="dji_sdk_read_cam" type="dji_sdk_read_cam" name="dji_sdk_read_cam" output="screen">     
   
	<param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="video_device" value="/dev/video0" />      
    <param name="pixel_format" value="yuyv" />
     
    <param name="camera_name" value="/camera/image_raw" />
    <param name="camera_info_url" value="package://visp_auto_tracker/models/calibration.ini" type="string" />
  </node>

</launch>
Then I roslaunch visp_auto_tracker tracklive_usb.launch ,here are always some problems like:
... logging to /home/ubuntu/.ros/log/fb150e98-36e3-11e6-a395-1dc4271d6692/roslaunch-tegra-ubuntu-7507.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:44260/

SUMMARY
========

PARAMETERS
 * /dji_sdk_read_cam/camera_info_url: package://visp_au...
 * /dji_sdk_read_cam/camera_name: /camera/image_raw
 * /dji_sdk_read_cam/image_height: 480
 * /dji_sdk_read_cam/image_width: 640
 * /dji_sdk_read_cam/pixel_format: yuyv
 * /dji_sdk_read_cam/video_device: /dev/video0
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /visp_auto_tracker/debug_display: True
 * /visp_auto_tracker/model_name: pattern
 * /visp_auto_tracker/model_path: /home/ubuntu/dutR...

NODES
  /
    dji_sdk_read_cam (dji_sdk_read_cam/dji_sdk_read_cam)
    visp_auto_tracker (visp_auto_tracker/visp_auto_tracker)

auto-starting new master
process[master]: started with pid [7518]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fb150e98-36e3-11e6-a395-1dc4271d6692
process[rosout-1]: started with pid [7531]
started core service [/rosout]
/opt/ros/indigo/lib/python2.7/dist-packages/roslib/packages.py:447: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  if resource_name in files:
process[visp_auto_tracker-2]: started with pid [7548]
process[dji_sdk_read_cam-3]: started with pid [7549]
0
[ INFO] [1466426349.298328831]: model full path=/home/ubuntu/dutRacing/src/vision_visp/visp_auto_tracker/models/pattern
[ INFO] [1466426349.302621247]: Model content=#VRML V2.0 utf8

DEF fst_0 Group {
children [

# Object "cube"
Shape {

geometry DEF cube IndexedFaceSet {

coord Coordinate { 
point [
-0.0765  -0.0765	0.000
 0.0765  -0.0765	0.000
 0.0765  0.0765	0.000
-0.0765  0.0765	0.000
-0.03825  -0.03825	0.000   
 0.03825  -0.03825	0.000
 0.03825  0.03825	0.000
-0.03825  0.03825  	0.000 
 ]
}

coordIndex [
 0,1,2,3,-1,
 4,5,6,7,-1,
]}
}

]
}

starting tracker
 *********** Parsing XML for Mb Edge Tracker ************ 
ecm : mask : size : 5
ecm : mask : nb_mask : 180
ecm : range : tracking : 10
ecm : contrast : threshold 5000
ecm : contrast : mu1 0.5
ecm : contrast : mu2 0.5
sample : sample_step : 4
sample : n_total_sample : 250
klt : Mask Border : 0
klt : Max Features : 10000
klt : Windows Size : 5
klt : Quality : 0.05
klt : Min Distance : 20
klt : Harris Parameter : 0.01
klt : Block Size : 3
klt : Pyramid Levels : 3
face : Angle Appear : 75
face : Angle Disappear : 75
camera : u0 : 192 (default)
camera : v0 : 144 (default)
camera : px : 600 (default)
camera : py : 600 (default)
lod : use lod : 0 (default)
lod : min line length threshold : 50 (default)
lod : min polygon area threshold : 2500 (default)
(L0) !!	/ViSP/src/tracking/mbt/vpMbTracker.cpp: loadVRMLModel(#1115) : coin not detected with ViSP, cannot load model : /home/ubuntu/dutRacing/src/vision_visp/visp_auto_tracker/models/pattern.wrl
terminate called after throwing an instance of 'vpException'
  what():  coin not detected with ViSP, cannot load model
[visp_auto_tracker-2] process has died [pid 7548, exit code -6, cmd /home/ubuntu/dutRacing/devel/lib/visp_auto_tracker/visp_auto_tracker /visp_auto_tracker/camera_info:=/dji_sdk/camera_info /visp_auto_tracker/image_raw:=/dji_sdk/image_raw __name:=visp_auto_tracker __log:=/home/ubuntu/.ros/log/fb150e98-36e3-11e6-a395-1dc4271d6692/visp_auto_tracker-2.log].
log file: /home/ubuntu/.ros/log/fb150e98-36e3-11e6-a395-1dc4271d6692/visp_auto_tracker-2*.log

in addition ，I install ViSP from source

wanlong0827, 2016-06-20:
I modify tracklive_usb.launch：

 
  <node pkg="visp_auto_tracker"type="visp_auto_tracker"name="visp_auto_tracker"output="screen">
    <param name="model_path" value="$(find visp_auto_tracker)/models" />
    <param name="model_name" value="pattern" />
    <param name="debug_display" value="True" />      
    <remap from="/visp_auto_tracker/camera_info" to="/dji_sdk/camera_info"/>
    <remap from="/visp_auto_tracker/image_raw" to="/dji_sdk/image_raw"/>
  </node>
  
  <node pkg="dji_sdk_read_cam" type="dji_sdk_read_cam" name="dji_sdk_read_cam" output="screen">     
   
	<param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="video_device" value="/dev/video0" />      
    <param name="pixel_format" value="yuyv" />
     
    <param name="camera_name" value="/camera/image_raw" />
    <param name="camera_info_url" value="package://visp_auto_tracker/models/calibration.ini" type="string" />
  </node>
</launch>

fspindle, 2016-06-22:
You define the model of the cube you wnt to track in vrml. This format is only supported if ViSP is build with Coin3D support.

- install Coin3D: ```sudo apt-get install libcoin80-dev```
- build/install ViSP again 
- build/install visp_auto_tracker again

fspindle, 2017-02-02:
Closed.
Reopen if the issue remains.

closed 2017-02-02 by fspindle