lagadic/vision_visp #55: visp_auto_tracker crashes on detection

agjunyent, 2016-07-18:
I'm trying to start using visp_auto_tracker to track a simple QR code (the one in the wiki). I can start it using `roslaunch visp_auto_tracker tracklive_usb.launch`, everything loads correct and I can see a window with the video of /image_raw.

The problem is that when I show the QR code to the webcam, all crashes with the following error:
```
[visp_auto_tracker-2] process has died [pid 12529, exit code -11, cmd /home/albert/catkin_ws/devel/lib/visp_auto_tracker/visp_auto_tracker /visp_auto_tracker/camera_info:=/usb_cam/camera_info /visp_auto_tracker/image_raw:=/usb_cam/image_raw __name:=visp_auto_tracker __log:=/home/albert/.ros/log/a987a6aa-4cee-11e6-87fa-b8ee651e75ac/visp_auto_tracker-2.log].
log file: /home/albert/.ros/log/a987a6aa-4cee-11e6-87fa-b8ee651e75ac/visp_auto_tracker-2*.log
```

Any help?

fspindle, 2016-08-16:
No idea. Are your `camera_info` parameters right. Check if the principal point is approximatly in the center of the image.

agjunyent, 2016-08-22:
Sorry for the delay, been in holidays...

I'll try this week and say you something.

Thanks!!

closed 2017-02-02 by fspindle