lagadic/vision_visp #57: bug report: visp_auto_tracker/src/node.cpp:135:40: error: ‘vpDetectorQRCode’ does not name a type

YuehChuan, 2017-01-07:
Environment:  x86_64  ubuntu16.04 ROS kinectic
I'll try to fix it.

```
[ 97%] Linking CXX shared library /home/rapyuta/rapyuta/rapyuta_ws/devel/lib/libauto_tracker.so
[ 97%] Built target auto_tracker
[ 97%] Building CXX object vision_visp/visp_auto_tracker/CMakeFiles/visp_auto_tracker.dir/src/node.cpp.o
[ 98%] Building CXX object vision_visp/visp_auto_tracker/CMakeFiles/visp_auto_tracker.dir/src/names.cpp.o
[ 98%] Building CXX object vision_visp/visp_auto_tracker/CMakeFiles/visp_auto_tracker.dir/src/main.cpp.o
/home/rapyuta/rapyuta/rapyuta_ws/src/vision_visp/visp_auto_tracker/src/node.cpp: In member function ‘void visp_auto_tracker::Node::spin()’:
/home/rapyuta/rapyuta/rapyuta_ws/src/vision_visp/visp_auto_tracker/src/node.cpp:135:40: error: ‘vpDetectorQRCode’ does not name a type
                         detector = new vpDetectorQRCode;
                                        ^
/home/rapyuta/rapyuta/rapyuta_ws/src/vision_visp/visp_auto_tracker/src/node.cpp:137:40: error: ‘vpDetectorDataMatrixCode’ does not name a type
                         detector = new vpDetectorDataMatrixCode;
                                        ^
vision_visp/visp_auto_tracker/CMakeFiles/visp_auto_tracker.dir/build.make:110: recipe for target 'vision_visp/visp_auto_tracker/CMakeFiles/visp_auto_tracker.dir/src/node.cpp.o' failed
make[2]: *** [vision_visp/visp_auto_tracker/CMakeFiles/visp_auto_tracker.dir/src/node.cpp.o] Error 1
CMakeFiles/Makefile2:5702: recipe for target 'vision_visp/visp_auto_tracker/CMakeFiles/visp_auto_tracker.dir/all' failed
make[1]: *** [vision_visp/visp_auto_tracker/CMakeFiles/visp_auto_tracker.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

```

YuehChuan, 2017-01-07:
This is on master branch

fspindle, 2017-01-10:
I suspect that you are building with an old ViSP version (< 2.10.0) or that you have multiple ViSP installions that bring a mess.

Could you check the ViSP versions you have on your computer:

```
$ sudo updatedb
$ locate vpConfig.h
```

and the one that is used by catkin:

```
$ cd catkin_wk/build
$ grep VISP_DIR *
```

YuehChuan, 2017-01-10:
Oh thanks for your reply.  
yes, originally I manually installed visp-snapshot-2017-01-07 .

`$ locate vpConfig.h`

Output:
```
/home/rapyuta/soft/visp-build/include/visp/vpConfig.h
/home/rapyuta/soft/visp-build/include/visp3/core/vpConfig.h
/home/rapyuta/soft/visp-snapshot-2017-01-07/cmake/templates/vpConfig.h.in
/usr/local/include/visp
/usr/local/include/visp3/core/vpConfig.h
```
I guess `/usr/local/include/visp` ` /usr/local/include/visp3/core/vpConfig.h` is the same as 
`/home/rapyuta/soft/visp-build/include/visp/vpConfig.h` `/home/rapyuta/soft/visp-build/include/visp3/core/vpConfig.h`?  just when we did `sudo make install` 
So  the issue is version conflict?

`$rapyuta_ws/build$ grep VISP_DIR *`
Output:
```
CMakeCache.txt:VISP_DIR:PATH=/usr/local/lib/cmake/visp
grep: CMakeFiles: Is a directory
grep: apriltags: Is a directory
grep: catkin: Is a directory
grep: catkin_generated: Is a directory
grep: epnp: Is a directory
grep: gtest: Is a directory
grep: test_results: Is a directory
grep: vision_visp: Is a directory
```
Thanks for your time.

YuehChuan, 2017-01-10:
The Visp Version is 3.0.1

fspindle, 2017-01-10:
Just to be sure, do you install libzbar-dev ?
```
$ sudo apt-get install libzbar-dev
```
If it was not installed, you have to enter in ViSP build tree, configure with cmake again and do a fresh make install

YuehChuan, 2017-01-10:
WOW! You are genius, LOVE YOU!

[image]
Yes, miss `$ sudo apt-get install libzbar-dev` and fresh install again.
I should read this carefully. http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-ubuntu.html
 Anyway thanks for your patient and time. YOLO!

closed 2017-01-10 by fspindle

TokyoClod, 2018-04-05:
Fixed with
`sudo apt-get install libzbar-dev libdmtx-dev`
and rebuild, reinstall visp, then use catkin_make to make the ros workspace