lagadic/vision_visp #66: visp_tracker (tracker_viewer) doesn't work with klt

ghanimmukhtar, 2017-07-28:
Hi there,

So I was testing the visp_tracker, when I use the default method (mbt) everything work, but when I pass through the launch file the parameter tracker_type as klt as follows:
```
 <node pkg="nodelet" type="nodelet" name="visp_tracker_client" output="screen"
        args="load visp_tracker/TrackerClient $(arg manager)">
    <param name="model_path" value="package://visp_tracker/models" />
    <param name="tracker_type" value="klt" />
    <param name="model_name" value="laas-box" />

    <!-- Load recommended settings for tracking initialization. They
      will be automatically forwarded to the tracking node if the
      initialization succeed. -->
    <rosparam file="$(find visp_tracker)/models/laas-box/tracker.yaml" />
  </node>
```

The node get stuck in the waitForImage() method with the periodic message:
```
waiting for a rectified image...
```

Any idea why? A quick look to the code shows that in both cases the image_subscriber is subscribing the same topic, so I don't really see why this is happening??
Thanks in advance for your feedback
Cheers!

fspindle, 2017-07-31:
It means maybe that the tracker doesn't have enough features (keypoints from klt). Try to reduce the value of the `klt > quality` parameter to get more features (has to be done in the xml file).

I recommend to rather use `mbt` or `mbt+klt` instead of just `klt`

ghanimmukhtar, 2017-07-31:
@fspindle Thanx for your reply, so now the klt is working, I am trying with klt ->quality = 0.015 and it starts to work. I will try to tune further and see how it goes, sadly I don't know exactly the the meaning of this parameter cause it is not explained in [visp tutorial](http://visp-doc.inria.fr/doxygen/visp-2.9.0/tutorial-tracking-mb.html) so I guess I will do it by try/error

fspindle, 2017-07-31:
Quality parameter meaning is described [here](http://visp-doc.inria.fr/doxygen/visp-daily/classvpKltOpencv.html#a1fb53a5f47b8ac62e9f6db586b670fbf).

ghanimmukhtar, 2017-08-01:
So I did the catkin_make with the release flag for visp, visp_bridge and visp_tracker, and I put the ```klt > quality``` parameters with 0.001 and as you can see in [this video](https://drive.google.com/open?id=0B8ioqiBiw2SSWjhHQTFIVUlWYjg), the first part is working well, then when I define the object and try to initialize the tracker it actually doesn't work still waiting for rectified image!! any ideas why the second part isn't working? I suspected that in the launch file I need to pass the parameter ```tracker_type``` as ```klt``` but it didn't work also. 
Sorry for the confusion, yesterday I thought it was working but actually I only checked the first part without initializing the tracker ;(

ghanimmukhtar, 2017-08-01:
Actually it worked now I had to add the parameter ```tracker_type``` to ```nodelet_manager```, ```tracker```, ```tracker_client``` besides the ```tracker_viewer```, my bad :)