lagadic/vision_visp #67: very sensitive tracking

ghanimmukhtar, 2017-07-28:
So I did one test with kinect v2, I made some changes in the launch file, and the trackers codes, besides adding my own model to track. You can see all those modifications [here](https://github.com/ghanimmukhtar/vision_visp) then I test the whole thing and it starts fine but after moving the object around the tracker start to get lost and then it crashes, it was rather fast, you can check [the video](https://drive.google.com/open?id=0B8ioqiBiw2SSNndtMDdRdWpsSUE) to asses the performance.

So my question is that normal? Or I need to tune some parameters to have better and robust tracking?
Cheers!

fspindle, 2017-07-31:
Since your object is textured, try with the hybrid tracker: mbt+klt has to be used in the launch file.
If you don't see keypoints in the viewer, try to modify the quality parameter. To tune the parameters you can use the dynamic reconfigure [tool](http://wiki.ros.org/visp_tracker/Tutorials/Fine%20tuning%20of%20the%20tracking%20parameters)

Concerning the frequency, check if everything (visp, visp_bridge and visp_tracker) is built in release: `catkin_make -DCMAKE_BUILD_TYPE=Release`

ghanimmukhtar, 2017-07-31:
So I tested it with klt thanx to your answer in #66 and it seems more robust but still after a while the tracker starts to get lost and can't correct itself. I will try to show by another video tomorrow.
All packages including visp v3 is build with the Release flag.

fspindle, 2017-07-31:
You may also try to modify the `me > range` parameter. Its value in pixel (typically 8) is used to search the edge in the next image from the previous position. If between 2 successive images your motion is near 20 pixels, the range value should be set to 25 (20+something).