lagadic/vision_visp #69: vision_visp build error

Juna2, 2017-09-06:
I use ubuntu 16.04 and ros kinetic.
I tried to install and build the whole vision_visp meta package
(I followed the instruction of http://wiki.ros.org/vision_visp)

And when I do
$ catkin_make -j4 -DCMAKE_BUILD_TYPE=Release
my computer can go any further after
[ 93%] Built target visp_hand2eye_calibration_client

It just stop. I waited for quite long time but It doesn't seem to get better.
Is there anything that I have to do more?

juna@juna-800G5M-800G5W:~/catkin_ws$ catkin_make -j4 -DCMAKE_BUILD_TYPE=Release
Base path: /home/juna/catkin_ws
Source space: /home/juna/catkin_ws/src
Build space: /home/juna/catkin_ws/build
Devel space: /home/juna/catkin_ws/devel
Install space: /home/juna/catkin_ws/install
####
#### Running command: "cmake /home/juna/catkin_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/juna/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/juna/catkin_ws/install -G Unix Makefiles" in "/home/juna/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/juna/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/juna/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/juna/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/juna/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.6
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 34 packages in topological order:
-- ~~  - dynamixel_driver
-- ~~  - dynamixel_motor (metapackage)
-- ~~  - dynamixel_tutorials
-- ~~  - hector_slam (metapackage)
-- ~~  - hector_slam_launch
-- ~~  - joy_teleop
-- ~~  - dynamixel_controllers
-- ~~  - mouse_teleop
-- ~~  - dynamixel_msgs
-- ~~  - key_teleop
-- ~~  - hector_map_tools
-- ~~  - hector_nav_msgs
-- ~~  - teleop_tools (metapackage)
-- ~~  - teleop_tools_msgs
-- ~~  - vision_visp (metapackage)
-- ~~  - dynamixel_sdk
-- ~~  - hector_geotiff
-- ~~  - hector_geotiff_plugins
-- ~~  - hector_marker_drawing
-- ~~  - ros_tutorials_service
-- ~~  - ros_tutorials_topic
-- ~~  - hector_compressed_map_transport
-- ~~  - rplidar_ros
-- ~~  - my_dynamixel_tutorial
-- ~~  - hector_imu_attitude_to_tf
-- ~~  - hector_imu_tools
-- ~~  - hector_map_server
-- ~~  - hector_trajectory_server
-- ~~  - hector_mapping
-- ~~  - visp_bridge
-- ~~  - visp_camera_calibration
-- ~~  - visp_hand2eye_calibration
-- ~~  - visp_tracker
-- ~~  - visp_auto_tracker
[... 319 lines cut ...]
[ 83%] Built target teleop_tools_msgs_generate_messages
[ 84%] Built target geotiff_node
[ 84%] Built target geotiff_saver
[ 84%] Built target hector_geotiff_plugins
[ 84%] Built target service_server2
[ 85%] Built target service_server
[ 85%] Built target ros_tutorials_service_generate_messages
[ 87%] Built target service_client
[ 87%] Built target topic_subscriber
[ 87%] Built target ros_tutorials_topic_generate_messages
[ 88%] Built target topic_publisher
[ 88%] Built target hector_mapping_generate_messages
[ 89%] Built target hector_mapping
[ 89%] Built target visp_camera_calibration_generate_messages
[ 90%] Built target visp_camera_calibration_calibrator
[ 91%] Built target visp_camera_calibration_image_processing
[ 91%] Built target visp_hand2eye_calibration_generate_messages
[ 92%] Built target visp_camera_calibration_camera
[ 92%] Built target visp_hand2eye_calibration_gencpp
[ 92%] Built target visp_tracker_gencpp
[ 92%] Built target visp_tracker_generate_messages
[ 92%] Built target visp_tracker_client
[ 92%] Built target tracker
[ 93%] Built target visp_hand2eye_calibration_calibrator
[ 93%] Built target visp_hand2eye_calibration_client

fspindle, 2017-09-10:
I don't see any reason why it hangs. It should be due to the compiler.
As workarround, you may try:
1. Don't use `-j4` but just `$ catkin_make`
2. If `1.` doesn't work, turn off `visp_hand2eye_calibration`build by
````
$ cd ~/catkin_ws/src/vision_visp/visp_hand2eye_calibration
$ touch CATKIN_IGNORE
$ catkin_make -j4
````

Akumar201, 2022-10-04:
Hello, I tried the following command to install visp `git clone -b kinetic-devel https://github.com/lagadic/vision_visp.git` , however, I am not sure that why I am getting this error, can anyone help me please to solve this error 

```
CMake Error at vision_visp/visp_bridge/CMakeLists.txt:16 (find_package):
  By not providing "FindVISP.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "VISP", but
  CMake did not find one.

  Could not find a package configuration file provided by "VISP" with any of
  the following names:

    VISPConfig.cmake
    visp-config.cmake

  Add the installation prefix of "VISP" to CMAKE_PREFIX_PATH or set
  "VISP_DIR" to a directory containing one of the above files.  If "VISP"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/akumar/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/akumar/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

```

fspindle, 2022-10-05:
You have to install ViSP:

1. either installing the binary package `ros-kinetic-visp` 
2. or building ViSP from source (see [quick install](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-ubuntu.html#install_ubuntu_quick)) and then adding `-DVISP_DIR=$VISP_WS/visp-build` to `catkin_make`

schefferac2020, 2023-01-11:
I'm running into the same issue (after running `sudo apt-get install ros-noetic-visp`). It looks like /opt/ros/noetic/visp only has a package.xml file?

fspindle, 2023-01-12:
Which ubuntu version?
Did you checkout `noetic` branch in `vision_visp` ?