lagadic/vision_visp #71: tracker run slow

marine0131, 2017-10-23:
Hi
i have successfully run the auto tracker, but it semm like run much slower than your example videos show in ros wiki.  I uploaded my video so you can see that:
[https://youtu.be/8PRvELsXeYI](url)

the frequency of  /image_raw is 20 Hz, but when i run the 

> rostopic  hz /visp_auto_tracker/code_messgae 

i got about 8Hz

and when I echo the /visp_auto_tracker/code_messgae I got (dose it strange?):
[image]

much thanks if you can give some suggestions!

fspindle, 2017-10-23:
Are you sure you build the tracker in Release:

    catkin_make -DCMAKE_BUILD_TYPE=Release

marine0131, 2017-10-23:
Hi
I just tried build the tracker in Release 

but it behaved exactly the same as before

fspindle, 2017-10-23:
The VM could explain the slow down.

marine0131, 2017-10-23:
I think so, when i decrease the image size to 320x240, the /code_message's publish frequency get 20 Hz,

but main issue now is the empty message in the /code_message, it still exist no matter how i change the image size
[image]

fspindle, 2017-10-23:
Are you sure that a message is encoded in the QRcode

marine0131, 2017-10-23:
I download the QRcode you provided in 
https://github.com/lagadic/visp_auto_tracker-deprecated/downloads

fspindle, 2017-10-23:
The code from https://github.com/lagadic/visp_auto_tracker-deprecated is deprecated. You should rather get https://github.com/lagadic/vision_visp that is a meta package that contains visp_auto_tracker.

marine0131, 2017-10-23:
sorry about the QRcode link,
 I cloned the right git stack, the  [https://github.com/lagadic/vision_visp](https://github.com/lagadic/vision_visp ), and checkout indigo-0.10.0 branch,    that is the newest version of indigo right?
[image]

marine0131, 2017-10-26:
I have resolve the problem @fspindle , I change the `ad-hoc-recovery`  from 1 to 0, then it works!

now I am a liitle confused about the  function of this `ad-hoc-recovery`. is there any detailed explain of these parameters?

`ad-hoc-recovery, ad-hoc-recovery-display, ad-hoc-recovery-ratio, ad-hoc-recovery-threshold, ad-hoc-recovery-size`