lagadic/vision_visp #75: Error process has died

project10ros, 2017-11-22:
This is the message I get:

[visp_auto_tracker-1] process has died [pid 3675, exit code -11, cmd /home/ruben/catkin_ws/devel/lib/visp_auto_tracker/visp_auto_tracker /visp_auto_tracker/camera_info:=/usb_cam/camera_info /visp_auto_tracker/image_raw:=/usb_cam/image_raw __name:=visp_auto_tracker __log:=/home/ruben/.ros/log/12c6b6d2-cf76-11e7-bf56-303a6428194b/visp_auto_tracker-1.log].
log file: /home/ruben/.ros/log/12c6b6d2-cf76-11e7-bf56-303a6428194b/visp_auto_tracker-1*.log

This massage appear when the camera see a QR-code and then stop running.
How can I solve this?

dohema, 2017-12-27:
i also meet this problem,have you solve this problem,can you give some advice,thank you

fspindle, 2017-12-27:
Which ROS version ?
Which ubuntu version ?
vision_visp from source or from ROS package ?
Does the tutorial work `roslaunch visp_auto_tracker tutorial.launch`?

dohema, 2017-12-28:
thank you for your reply,
     ROS distro : kinetic;
     System version : debian 9.0; 
     vision_visp  compile from source;
     the tutorial not work,roslaunch    visp_tracker    tutorial.launch also don't work
      in  roslaunch  visp_tracker tutorial.launch: the error is  [visp_tracker_client-1] process has died [ exit code -11, ....
      in  roslaunch  visp_auto_tracker tutorial.launch : the error is  [visp_auto_tracker-1] process has died [exit code -11, ...
      and i also install visp library from source ,hope to hear your back

GuillermoHerreraF, 2020-03-09:
Hello! I'm having the same problem when I run the visp_auto_tracker node with the raspicam_node. I re-mapped the topic names to match the ones of the raspicam_node. However, when I run the launch file the terminal throws this message:

[visp_auto_tracker-2] process has died [pid 1991, exit code -11, cmd /opt/ros/kinetic/lib/visp_auto_tracker/visp_auto_tracker /visp_auto_tracker/camera_info:=/raspicam_node/camera_info /visp_auto_tracker/image_raw:=/raspicam_node/image __name:=visp_auto_tracker __log:=/home/raspberry/.ros/log/0ca0ea18-61e9-11ea-b307-b827eb583008/visp_auto_tracker-2.log].
log file: /home/raspberry/.ros/log/0ca0ea18-61e9-11ea-b307-b827eb583008/visp_auto_tracker-2*.log

I'm running ROS kinetic on Ubuntu Mate 16 on my Raspiberry Pi 3 Model B. When I run the turorial from the visp_auto_tracker package it runs fine.