lagadic/vision_visp #79: Ball Detection!

cheikhnasy, 2018-02-19:
Hello  everyone ! 
Can someone please help me in the design of an init file to detect a ball? i have my cao file ready but my initialization fails .

Thank you !

fspindle, 2018-02-19:
I never tried to track a ball with the model-based tracker using the model of a circle. 
I'm not sure it will work. Could be interesting to test using only the edges. The hybrid approach with KLT cannot work since the keypoints will not lie on a plane (they are on a sphere).

The init file should contain the coordinates of 4 3D points used to compute the initial pose. The user has than to click on the corresponding points that are in the image. It the radius is 0.1 the init file could contain the position of the point located a 12:00, 9:00, 6:00, 3:00 (if I consider the ball as a watch). If X goes from 9:00 to 3:00, Y goes from 6:00 to 12:00 the init file should look like:

```
4 
 0    0.1  0
-0.1  0    0
 0   -0.1  0
 0.1  0    0
```
The difficulty for the user would be to click on the ball contour at point 12:00, 9:00, 6:00, 3:00.

Then the cao file:

```
V1
# Ball model
3                    # Number of points
# 3D points
0    0    0       # pt 0: center of the ball
0   -0.1  0       # pt 1: bottom
0.1  0    0       # pt 2: right
# 3D lines
0
# Faces from 3D lines
0
# Faces from 3D points
0
# 3D cylinders
0
# 3D circles
1              # Number of circles
0.1 0 1 2  # circle 0: [radius] [circle center] [bottom point] [right point]
```

cheikhnasy, 2018-02-19:
Thank you , i will give it a try . Also, i am used to work with meters,centimeters ....! are the dimensions in the cao file in inches ?(just to make sure )

fspindle, 2018-02-19:
In ViSP, everything in meter/radian

cheikhnasy, 2018-02-19:
Cool ! thank you !

cheikhnasy, 2018-02-19:
i just tested ! i have a successful initialization but i loose the tracking whenever i move the ball!

cheikhnasy, 2018-02-19:
[image]

cheikhnasy, 2018-02-19:
works !as long as i move the ball slowly ! can i change this aspect of the tracking? also thank you ,your package is great!

fspindle, 2018-02-19:
Try to increase the range parameter to allow larger displacement in the image between 2 successive frames.

wafaezouhair, 2018-04-18:
Hi cheikhnasy, 
please how did you replace the basic photo with your own photo in "VISP MBT tracker initilization",  also if you have a idea how to run in real time ..

Thank you