lagadic/vision_visp #83: visp_auto_tracker build issue

henryng375, 2018-06-26:
When I built the package, I got such error:
CMake Error at /opt/ros/indigo/share/roscpp/cmake/roscppConfig.cmake:141 (message):
  Project 'visp_bridge' tried to find library 'pthread'.  The library is
  neither a target nor built/installed properly.  Did you compile project
  'roscpp'? Did you find_package() it before the subdirectory containing its
  code is included?
Call Stack (most recent call first):
  vision_visp/visp_bridge/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/darwin/work/ros_ws/visp_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/darwin/work/ros_ws/visp_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

But, I didn't see any pthread package include in CMakelist.

fspindle, 2018-06-26:
Install `pthread`using 
```
sudo apt-get install libpthread-stubs0-dev
```

closed 2018-06-26 by fspindle

henryng375, 2018-06-26:
@fspindle I have already installed `pthread`. But, still get the same error.
Actually, when I built usb_cam, I also got the same error.

henryng375, 2018-06-28:
@fspindle Problem solved. Simply Reinstall Ubuntu and upgrade to 16.04