lagadic/vision_visp #87: Kinova camera calibration

GianlucaCerilli, 2018-11-05:
Hello,

I am using a Kinova Jaco2 arm with a monocular camera mounted on the end effector. Is it possible to use this package for the camera calibration? I am on Ubuntu 16.04 with ros kinetic.

Thank you very much

fspindle, 2018-11-07:
If you mean the hand-eye calibration I would say yes if you have a node that is able to publis the pose of the end-effector in the robot base frame as explained [here](http://wiki.ros.org/visp_hand2eye_calibration?distro=lunar#Nodes).

GianlucaCerilli, 2018-11-07:
HI @fspindle,

Thank you for your reply. 
I don't understand some steps, I am sorry, so I wanted to ask you few things if it is possible.

- For this code to work with my camera and robot, is it necessary to change only the topic parameters in ''client.cpp" code?

- I am using a chessboard as object for camera calibration. How do you specify in the code the origin of the frame of the object to both put it in a know position wrt the world frame and to also get the extrinsic parameters from the camera?

- I know the end-effector pose wrt the robot base frame. I should also know the position of the chessboard wrt the robot base frame. But I don't know the pose of the camera (mounted on the end-effector) wrt the object. So, how can you feed the camera-to-object transformations?

Sorry for the many questions and thank you very much!