lagadic/vision_visp #90: Build Error building vision_visp/visp_hand2eye_calibration

chandanlal92, 2019-03-18:


chandanlal92, 2019-03-18:
CMakeFiles/visp_hand2eye_calibration_calibrator.dir/src/calibrator.cpp.o: In function `visp_hand2eye_calibration::Calibrator::computeEffectorCameraCallback(visp_hand2eye_calibration::compute_effector_cameraRequest_<std::allocator<void> >&, visp_hand2eye_calibration::compute_effector_cameraResponse_<std::allocator<void> >&)':
calibrator.cpp:(.text+0x41b): undefined reference to `vpCalibration::calibrationTsai(std::vector<vpHomogeneousMatrix, std::allocator<vpHomogeneousMatrix> > const&, std::vector<vpHomogeneousMatrix, std::allocator<vpHomogeneousMatrix> > const&, vpHomogeneousMatrix&)'
CMakeFiles/visp_hand2eye_calibration_calibrator.dir/src/calibrator.cpp.o: In function `visp_hand2eye_calibration::Calibrator::computeEffectorCameraQuickCallback(visp_hand2eye_calibration::compute_effector_camera_quickRequest_<std::allocator<void> >&, visp_hand2eye_calibration::compute_effector_camera_quickResponse_<std::allocator<void> >&)':
calibrator.cpp:(.text+0x104e): undefined reference to `vpCalibration::calibrationTsai(std::vector<vpHomogeneousMatrix, std::allocator<vpHomogeneousMatrix> > const&, std::vector<vpHomogeneousMatrix, std::allocator<vpHomogeneousMatrix> > const&, vpHomogeneousMatrix&)'
collect2: error: ld returned 1 exit status
vision_visp-kinetic-devel/visp_hand2eye_calibration/CMakeFiles/visp_hand2eye_calibration_calibrator.dir/build.make:262: recipe for target '/home/chandan_main/catkin_ws/devel/lib/visp_hand2eye_calibration/visp_hand2eye_calibration_calibrator' failed
make[2]: *** [/home/chandan_main/catkin_ws/devel/lib/visp_hand2eye_calibration/visp_hand2eye_calibration_calibrator] Error 1
CMakeFiles/Makefile2:19647: recipe for target 'vision_visp-kinetic-devel/visp_hand2eye_calibration/CMakeFiles/visp_hand2eye_calibration_calibrator.dir/all' failed
make[1]: *** [vision_visp-kinetic-devel/visp_hand2eye_calibration/CMakeFiles/visp_hand2eye_calibration_calibrator.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

chandanlal92, 2019-03-18:
I am trying to install vision_visp to use easy_hand eye calibration.I am getting this build error

fspindle, 2019-03-19:
It means that you don't link vision_visp to ViSP 3.2.0 or more recent but with a previous version

chandanlal92, 2019-03-19:
I guess I have compiled with the latest algorithm anyway I will test it and again and let you know.

chandanlal92, 2019-03-21:
How to link vision_visp to ViSP 3.2.0?

fspindle, 2019-03-21:
Using `catkin_make -DVISP_DIR=<path>/visp-build` where `visp-build` is the folder corresponding to the parent dir that contains ViSP 3.2.0 build (see [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-ubuntu.html)).

If  you did `make install` in `visp-build` folder, you have to point `ViSP_DIR` to the installation folder  `-DVISP_DIR=/usr/local/lib/cmake/visp`. If this is the case remove all other ViSP installations (potentially the one done with `apt-get install libvisp-dev` or `ros-kinetic-visp` packages that bring ViSP 3.1.0) in `/usr/...` to be sure that you have only one.

chandanlal92, 2019-03-21:
Thank you It worked.

closed 2019-03-21 by chandanlal92