lagadic/vision_visp #94: vision_visp new version build error

Jakubach, 2019-06-15:
Hello, 
I occured an error while building updated package. I use ros melodic and ubuntu.

Command error returns:
````
error: no matching function for call to ‘vpMbGenericTracker::getLline(std::__cxx11::list<vpMbtDistanceLine*>&)’
         tracker_.getLline(linesList);
                                    ^
In file included from /home/jakub/catkin_ws/src/vision_visp/visp_tracker/src/libvisp_tracker/tracker-client.cpp:26:0:
/opt/ros/melodic/include/visp3/mbt/vpMbGenericTracker.h:136:16: note: candidate: virtual void vpMbGenericTracker::getLline(const string&, std::__cxx11::list<vpMbtDistanceLine*>&, unsigned int) const
   virtual void getLline(const std::string &cameraName, std::list<vpMbtDistanceLine *> &linesList,
                ^~~~~~~~
/opt/ros/melodic/include/visp3/mbt/vpMbGenericTracker.h:136:16: note:   candidate expects 3 arguments, 1 provided
/home/jakub/catkin_ws/src/vision_visp/visp_tracker/src/libvisp_tracker/tracker-client.cpp:420:36: error: no matching function for call to ‘vpMbGenericTracker::getLline(std::__cxx11::list<vpMbtDistanceLine*>&)’
         tracker_.getLline(linesList);
                                    ^
In file included from /home/jakub/catkin_ws/src/vision_visp/visp_tracker/src/libvisp_tracker/tracker-client.cpp:26:0:
/opt/ros/melodic/include/visp3/mbt/vpMbGenericTracker.h:136:16: note: candidate: virtual void vpMbGenericTracker::getLline(const string&, std::__cxx11::list<vpMbtDistanceLine*>&, unsigned int) const
   virtual void getLline(const std::string &cameraName, std::list<vpMbtDistanceLine *> &linesList,
                ^~~~~~~~
/opt/ros/melodic/include/visp3/mbt/vpMbGenericTracker.h:136:16: note:   candidate expects 3 arguments, 1 provided
/home/jakub/catkin_ws/src/vision_visp/visp_tracker/src/libvisp_tracker/tracker.cpp: In member function ‘void visp_tracker::Tracker::updateMovingEdgeSites(visp_tracker::MovingEdgeSitesPtr)’:
/home/jakub/catkin_ws/src/vision_visp/visp_tracker/src/libvisp_tracker/tracker.cpp:163:37: error: no matching function for call to ‘vpMbGenericTracker::getLline(std::__cxx11::list<vpMbtDistanceLine*>&, int)’
       tracker_.getLline(linesList, 0);
                                     ^
```
`

Jakubach, 2019-06-19:
Fixed by downloading ros-melodic-visp package from Synaptic.

closed 2019-06-19 by fspindle

AnukritiSinghh, 2021-12-19:
Hey can you please let me know how to solve this using synaptic?

Jakubach, 2021-12-20:
Hello, just install Synaptic if you do not have it:  Use Synaptic for more
advanced software management (ubuntu.com)
<https://help.ubuntu.com/stable/ubuntu-help/addremove-install-synaptic.html.en>
and then search for "ros-melodic-visp" in this app. Then it should install
correctly.

Good luck!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Wolny
od wirusów. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

AnukritiSinghh, 2021-12-20:
Thankyou. That worked!