lagadic/vision_visp #96: visp_auto_tracker: Overwritting object position's frame id

Pyrojambo, 2019-08-27:
What is the reason for overwriting the frame_id in the header of the PoseStamped message for in line 194 - 196 of node.cpp of the vision_auto_tracker?
```
ps.header = image_header_;
ps.header.frame_id = tracker_ref_frame;
object_pose_publisher.publish(ps);
```
The `tracker_ref_frame` is set to "\map" in names.cpp which overwrites whatever frame the image (and therefore the pose calculated) used as a reference. The same is also done to the covariance position.

fspindle, 2019-08-27:
To have pose and covariance corresponding to the processed image timestamp

Pyrojambo, 2019-08-28:
The timestamp is copied to the pose  with `ps.header = image_header_`.
My question is the next line (`ps.header.frame_id = tracker_ref_frame;`) where you overwrite the reference frame from the image's reference frame to "/map".
A quick example of what I mean is:
- A barcode is detected
- The pose of the barcode is calculated to be x=1, y=0, z=0 (ignore orientation for the example)
- The pose has its header (timestamp and reference frame) set to those of the image
- The pose is with respect to the camera i.e. the barcode is 1m away from the camera, in line with its centre
- The pose's reference frame is overwritten to be "/map", so now the barcode is said to be 1m away from the origin of the map
- If we have a robot at x=10, y=0, z=0 wrt "/map"; and give it the above pose as a goal it will travel 9m in the -x direction rather than 1m in the direction that the camera faces

fspindle, 2019-09-02:
Sorry, I miss read your first message.

I don't see any good reason to overwrite `frame_id`.

nickvaras, 2019-10-17:
#97