lagadic/vision_visp #99: Questions about number of samples in hand2eye_calibration

double344931987, 2019-12-30:
I use your code with realsense in Eye on Base calibration
During my experiment, if I took around 15 images, the results looks good but not exactly correct.
So, I take around 25, 30 and 40 samples, the results go worse.
The result is confused me, in my opinion, the more samples, more data, the results more accurate, but actually not.
Do you have this experience and know the problem is.
I am appreciate for your explanation firstly.

fspindle, 2020-02-04:
Usually I use between 8 and 12 images.

double344931987, 2020-02-04:
Thanks for your reply firstly.
However, I am also not very clearly about the answer of my problem.
If the working environment comes larger, it is inevitable to take more samples to cover the whole working environment. 12 images is not enough sometimes I believe. 
Could you please give me some ideas or possible reasons, it is OK I could try it myself and publish the results here to make the algorithm more robust.  
Thanks in advance.

FJarecki, 2020-02-10:
Hi,
I got similar results to yours double344931987. Did u find some solutions?

Jakubach, 2020-02-10:
Hello,
I think this is very interesting topic as accuracy of hand-eye calibration depends on many factors which should be carefully supervised. When you increase your working region you should also take it into account on your camera's internal calibration similarly increasing area of collected images.
Then in you should take care about characteristics of samples for hand-eye. There are some very important tips mentioned in Tsai's article which VISP_hand2eye_calibration implements, you can check it in section III C: [https://pdfs.semanticscholar.org/19b3/89a797a55c8b63dca8b6d1889df4cff8bfaa.pdf](url). 
Normally, you should minimalize translation between positions and maximize rotation and from my experience it works even for bigger number of samples. It gave me significant improvement of accuracy and I took about 140 samples. 

[image]
We can read about influence of number of samples that error is inversely proportional to the root of number of samples, so increasing number of correct samples should result in improving accuracy up to certain point.
There is still one more thing which on I am thinking on, from article's figure's 6 label:
> Pairs of stations should be selected such that the interstation angle is as large as possible, and the angle between different interstation rotation axes are as large as possible

I think, why there is a star shape there?
Try to take into account this tips and give a feedback if you got satisfying results :)

double344931987, 2020-03-04:
Interesting, thanks for your reply.
By the way, Is there any method that I could got the error of re project?
I think it is good way to know the accuracy.

Jakubach, 2020-04-08:
Hello, sorry for long time break but I've had to be sure to answer.
Here is comparasion between calibration with 54 samples vs 189 samples.

Mean residual rMo(54) - translation (m) = 0.000119247
Mean residual rMo(54) - global = 0.000364767

Mean residual rMo(189) - translation (m) = 0.000122502
Mean residual rMo(189) - global = 0.000308128

In terms of accuracy it gave me around 0,01 mm improvement by axes but I tested in Gazebo, where errors scale in hand-eye system is small. Visp hand2eye_calibration return this parameter, the first one takes only translation error and the second sums rotation and translation parts so I find it as a good indicator for testing accuracy.

fspindle, 2020-04-08:
I think that the residual values are good indicators.

Don't forget that the quality of the estimation depends on the quality of your input data.
I think that using between 10 and 20 samples could lead to better results than using 100 or more if the samples are well covering the half sphere over the calibration grid. 

See https://github.com/lagadic/visp/issues/729

double344931987, 2020-04-09:
Thanks all of the comments. I will try and give the feedback.