lagadic/vision_visp #101: How can I run a pose estimation from 4 planar points with ROS

phamhoanganhbk, 2020-02-03:
Hi,
I have seen that ViSP has supported with ROS with some examples: visp_auto_tracker with QR Code... But it seems that not all tutorials have been converted to ROS. For example, i would like to run an example : "Pose estimation from 4 planar points"
[https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-pose-estimation.html](url)
How can I run this example with ROS ?
Thank you very much for your help,

fspindle, 2020-02-04:
You have to write your own ROS node inspired from [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-pose-estimation.html) and from existing [visp_blob_tracker](https://github.com/lagadic/visp_blob_tracker) ROS package

phamhoanganhbk, 2020-02-04:
Thank you very much for your reply.

closed 2020-12-11 by phamhoanganhbk