lagadic/vision_visp #104: QR-code reading range problem

Rojekk, 2020-08-06:
Hi,

visp_auto_tracker works great but i want to able to read QR codes from 2 meters distance. The example QR code in wiki page's size is 5x5 and i can track it from 55 cm at most. When you go for more distance it keeps tracking but starting detection point is important for me. I keep using larger and larger QR codes but with a QR code with 20x20 cm size i got 190 cm range which is still lower than 200 and i can't go any bigger on QR codes, i'm limited with A4 paper size. 

Any suggestions on tracking Qr codes from long distances ?

fspindle, 2020-08-24:
I would suggest to use rather AprilTags instead of QR code.
I didn't find time to update `visp_auto_tracker` with AprilTags, but starting from this [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-detection-apriltag.html) it should be not so difficult.

EtienneAr, 2022-01-07:
Hello,

I wrote a little piece of code to include the April tags detector in visp_auto_tracker. I don't know if it's what you want @Canbjerg.

I opened a PR #117 because this feature interest me also.