lagadic/vision_visp #105: Intel RealSense D435i with ROS package 'visp_auto_tracker': Pixel formats error !!!

datvtmkac, 2020-09-15:
Hi everyone,

I'm currently working with ROS package 'visp_auto_tracker'. I have done testing this pkg with some USB-Webcam (2D Logitech C270) and it worked really well. I've tried to use a different high quality camera, especially the Intel RealSense D435i for this stuff but I got these following issues:
1. These are logs: unknown control 'white_balance_temperature_auto'; 'focus_auto' in terminal console
2. Issue with Pixel formats of the D435i camera

I tried to fix it this by changing the node config in "tracklive_usb.launch" file. For usb camera acquisition node, I changed all available type of "pixel_format" parameter in this package but I expected RGB image mode did not come. Following the "usb_cam" node, there are 6 types of value available: 'yuyv', 'uyvy', 'mjpeg', 'yuvmono10', 'rgb24' and 'grey'.

Below here is content of "tracklive_usb.launch" file:

`
<?xml version='1.0' ?>
<launch>	
  
  <node pkg="visp_auto_tracker" type="visp_auto_tracker" name="visp_auto_tracker" output="screen">
    <param name="model_path" value="$(find visp_auto_tracker)/models" />
    <param name="model_name" value="pattern" />
    <param name="debug_display" value="True" />
	      
    <remap from="/visp_auto_tracker/camera_info" to="/usb_cam/camera_info"/>
    <remap from="/visp_auto_tracker/image_raw" to="/usb_cam/image_raw"/>
  </node>

  
  <node pkg="usb_cam" type="usb_cam_node" name="usb_cam" output="screen">      
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />

    <param name="video_device" value="/dev/video3" />      
    <param name="pixel_format" value="yuyv" />
     
    <param name="camera_name" value="/camera/image_raw" />
    <param name="camera_info_url" value="package://visp_auto_tracker/models/calibration.ini" type="string" />
  </node>

</launch>`

And here are my following issues, I grab it in screenshots, sorry for this inconvenient:
[image]
[image]
[image]

So, I think my issues are about pixel formats of this Intel D435i camera, but I haven't solved it yet by myself. I hope you guys can give me a hand! Any help are very much appreciated!

Thanks,
datvt.

datvtmkac, 2020-09-18:
I solved it. My mistake, sorry! I just used depth module instead RGB module. Now everything is fine!

closed 2020-09-18 by datvtmkac

Gowresh-MARC, 2021-03-26:
Can you please share your launch file to use auto_tracker with D435i? I am trying to do the same, but not able to successfully.