lagadic/vision_visp #106: visp_hand2eye_calibration:  Problem in solving Hand-Eye Calibration by VVS

aycameric, 2020-12-08:
Hello, 
I am using visp_hand2eye_calibration and have a few questions:
1. if I take six subsequent samples from my endeffector position, then there will not be much difference, since it takes a while with my spline cartesian controller and you said it's better to have a rotation in the poses, so I am saving six values with a user request. Is this right?
2. I get this error:  Problem in solving Hand-Eye Calibration by VVS
after I execute the calibration node.

Thank you in advance!
Best regards,

aycameric, 2020-12-09:
And when I use the service I get this error:

Problem in solving Hand-Eye Rotation by Procrustes method 

Why is that?

fspindle, 2020-12-09:
It means that with your data the algorithm is unable to converge.

This [image](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-calibration-extrinsic.html#calib_ext_acq_data) and the corresponding [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-calibration-extrinsic.html) show what kind of good poses should be acquired to succeed.

closed 2020-12-15 by fspindle

aycameric, 2020-12-21:
Does the algorithm use the AX=XB problem? And can I use it for eye-to-hand calibration instead of eye-in-hand? So I have a static camera position.. I solved the previous issue with better data but my matrix that I got is not correct.