lagadic/vision_visp #110: Error while running ViSP moving edge tutorial

gy2256, 2021-02-06:
Operating System: Ubuntu 20.04
ROS: Noetic

I have compiled both visp and vision_visp from source without any problems. Then, I tried to follow the tutorial [https://wiki.ros.org/visp_tracker/Tutorials/Getting%20started](https://wiki.ros.org/visp_tracker/Tutorials/Getting%20started), and encountered the following issue after running `roslaunch visp_tracker tutorial.launch`. Here is the error message:

```
SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /tf_localization/object_translation_qw: 0.87906866
 * /tf_localization/object_translation_qx: 0.04655744
 * /tf_localization/object_translation_qy: -0.12974845
 * /tf_localization/object_translation_qz: -0.45632887
 * /tf_localization/object_translation_x: 1.84421063
 * /tf_localization/object_translation_y: -0.00836844
 * /tf_localization/object_translation_z: 0.52310595
 * /tracker_mbt/camera_prefix: /wide_left/camera
 * /tracker_mbt/tracker_type: mbt+klt
 * /tracker_mbt_client/frame_size: 0.2
 * /tracker_mbt_client/mask_size: 3
 * /tracker_mbt_client/model_name: laas-box
 * /tracker_mbt_client/model_path: package://visp_tr...
 * /tracker_mbt_client/mu1: 0.5
 * /tracker_mbt_client/mu2: 0.5
 * /tracker_mbt_client/n_mask: 180
 * /tracker_mbt_client/ntotal_sample: 800
 * /tracker_mbt_client/range: 10
 * /tracker_mbt_client/sample_step: 3.0
 * /tracker_mbt_client/threshold: 2000.0
 * /tracker_mbt_client/tracker_type: mbt+klt
 * /tracker_mbt_viewer/frame_size: 0.2
 * /tracker_mbt_viewer/tracker_name: tracker_mbt
 * /use_sim_time: False

NODES
  /
    base_link_to_camera (tf/static_transform_publisher)
    rosbag (rosbag/play)
    tf_localization (visp_tracker/tf_localization.py)
    tracker_mbt (visp_tracker/tracker)
    tracker_mbt_client (visp_tracker/visp_tracker_client)
    tracker_mbt_viewer (visp_tracker/visp_tracker_viewer)

auto-starting new master
process[master]: started with pid [46171]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 29e3f0d2-6884-11eb-84fc-495972883b37
process[rosout-1]: started with pid [46181]
started core service [/rosout]
process[rosbag-2]: started with pid [46188]
process[tracker_mbt-3]: started with pid [46189]
process[tracker_mbt_client-4]: started with pid [46190]
process[tracker_mbt_viewer-5]: started with pid [46191]
process[base_link_to_camera-6]: started with pid [46197]
process[tf_localization-7]: started with pid [46198]
[ WARN] [1612620242.099435495]: the camera_prefix parameter does not exist.
This may mean that:
- the tracker is not launched,
- the tracker and viewer are not running in the same namespace.
Traceback (most recent call last):
  File "/home/du/catkin_ws/src/vision_visp/visp_tracker/scripts/tf_localization.py", line 21, in <module>
    import tf
  File "/opt/ros/noetic/lib/python3/dist-packages/tf/__init__.py", line 30, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/noetic/lib/python3/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/noetic/lib/python3/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define init function (init_tf2)
[tf_localization-7] process has died [pid 46198, exit code 1, cmd /home/du/catkin_ws/src/vision_visp/visp_tracker/scripts/tf_localization.py __name:=tf_localization __log:=/home/du/.ros/log/29e3f0d2-6884-11eb-84fc-495972883b37/tf_localization-7.log].
log file: /home/du/.ros/log/29e3f0d2-6884-11eb-84fc-495972883b37/tf_localization-7*.log
[ WARN] [1612620243.155923060]: [Node: /tracker_mbt_viewer]
The model_description parameter does not exist.
This may mean that:
- the tracker is not launched or not initialized,
- the tracker and viewer are not running in the same namespace.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to load the model "package://visp_tracker/models/laas-box/laas-box"
Do you use resource_retriever syntax?
I.e. replace /my/model/path by file:///my/model/path
[tracker_mbt_client-4] process has died [pid 46190, exit code -6, cmd /home/du/catkin_ws/devel/lib/visp_tracker/visp_tracker_client __name:=tracker_mbt_client __log:=/home/du/.ros/log/29e3f0d2-6884-11eb-84fc-495972883b37/tracker_mbt_client-4.log].
log file: /home/du/.ros/log/29e3f0d2-6884-11eb-84fc-495972883b37/tracker_mbt_client-4*.log
^C[base_link_to_camera-6] killing on exit
[tracker_mbt_viewer-5] killing on exit
[tracker_mbt-3] killing on exit
[rosbag-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

I guess it's a mix use of python2 and python3 issue?

fspindle, 2021-02-08:
Fixed in `neotic` and  `neotic-devel` branches using `python3`.

- I can't merge the changes in `master` branch since changes are not working with `melodic`.
- There remain a lot of warnings launching `visp_tracker` tutorial that I can't figure out
    ```
    Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame object_position at time 1312292862.067297 according to authority /tracker_mbt
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
    Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame object_position at time 1312292862.067297 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
    ```

gy2256, 2021-02-14:
I have also solved the issue by using the following command in Ubuntu 20.04:
`sudo apt install python-is-python3`

closed 2021-02-14 by gy2256