lagadic/vision_visp #112: Vision Visp noetic

ranjitkathiriya, 2021-03-02:
Hello there,

I am compiling vision_visp into noetic and I am getting this error.

```
Errors     << visp_tracker:cmake /home/kuka-ai/ws_moveit/logs/visp_tracker/build.cmake.000.log
CMake Error at /home/kuka-ai/ws_moveit/src/vision_visp/visp_tracker/CMakeLists.txt:23 (find_package):
  By not providing "FindVISP.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "VISP", but
  CMake did not find one.

  Could not find a package configuration file provided by "VISP" with any of
  the following names:

    VISPConfig.cmake
    visp-config.cmake

  Add the installation prefix of "VISP" to CMAKE_PREFIX_PATH or set
  "VISP_DIR" to a directory containing one of the above files.  If "VISP"
  provides a separate development package or SDK, be sure it has been
  installed.

cd /home/kuka-ai/ws_moveit/build/visp_tracker; catkin build --get-env visp_tracker | catkin env -si  /usr/bin/cmake /home/kuka-ai/ws_moveit/src/vision_visp/visp_tracker --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/kuka-ai/ws_moveit/devel/.private/visp_tracker -DCMAKE_INSTALL_PREFIX=/home/kuka-ai/ws_moveit/install -DCMAKE_BUILD_TYPE=Release; cd -
...............................................................................
```

Any idea how to fix it?

Thanks and regards,
Ranjit

tonydle, 2021-03-07:
Hi @ranjitkathiriya, I also ran into the same issue. I think it's because the package uses VISP, and it doesn't officially support noetic yet. I tried using rosdep as guided on http://wiki.ros.org/vision_visp, and it also didn't work for the same reason.
What worked for me was building VISP from source, following this: https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-ubuntu.html

I hope that helped!

ranjitkathiriya, 2021-03-09:
Thanks, @tonydle,

Will check and let you know.

ranjitkathiriya, 2021-03-10:
Thanks, it is working fine.

closed 2021-03-10 by ranjitkathiriya