lagadic/vision_visp #114: Tracked object position covariance matrix for visp_tracker

dvedova, 2021-04-19:
Hi, when I am running the visp_tracker on the object I would like to track, I initialize the tracker, and track it, and it works fine. 

But when I go in and plot the diagonal elements of the covariance matrix, I get incredibly small values (1e-6 m^2). When I square root this number to get the value in meters, it's equal to 0.001, or 1mm. Does it make sense that the tracker is that confident in the tracked position of my object?

I am using a simulated camera image from coppeliasim, not a real camera image so maybe it's just that my simulated camera isn't very noisy?

fspindle, 2021-04-19:
The behavior you describe is truth.
You need to know that the covariance matrix is not a good indicator of the tracking quality.
Even when the tracker start to diverge, the diagonal doesn't increase so much since the tracker converge to a small error corresponding to something that may be completely wrong.

If your goal is to analyse the covariance to get an idea of the tracker confidence, I suggest to use rather the [projection error indicator](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-tracking-mb-generic.html#mb_generic_advanced_failure_detection).

dvedova, 2021-04-19:
Thank you for your reply. I'm not sure I completely understand this projection error. The ViSP tracker doesn't necessarily have access to the ground truth object pose, so how does the projection error calculation function know what to measure the error with respect to? 

My goal is to understand how confident the tracker is in a given measurement at a given time.

fspindle, 2021-04-19:
The ground truth is not used.

Given the estimed pose, we can project each line of the model in the image. The normal in the image of this projected line is then compared to the gradient along the projected line. When they are aligned the indicator is near 0 (angle = 0). When the angle increases, it means that there is a projection error (ie a tracking error). You can be confident in the results when the indicator has a small value.

dvedova, 2021-04-19:
Okay great. Thank you!

dvedova, 2021-04-20:
One additional question: What is the intended purpose of the object_position_covariance topic? Is it meant to be a covariance matrix that can be used for filtering purposes (e.g. Kalman filters)?

fspindle, 2021-04-21:
Yes, this was the original purpose.

dvedova, 2021-04-21:
Okay great. But based on what you said above that the covariance isn't very reliable, would you recommend using the object_position_covariance in a kalman filter?

fspindle, 2021-04-21:
We never tried, but feel free to try

dvedova, 2021-04-21:
Sounds good. Thanks for the replies!

closed 2021-04-21 by dvedova