lagadic/vision_visp #123: catkin_make error on Noetic

jifu23, 2022-10-07:
Hi! I was trying to build visp_ros from source, then I need to build visp-bridge, but I got these error. Could you please help me with that? Thanks!
`catkin_make -DCMAKE_BUILD_TYPE=Release --pkg visp_ros
Base path: /home/jifu/ros_workspaces/visp_ws
Source space: /home/jifu/ros_workspaces/visp_ws/src
Build space: /home/jifu/ros_workspaces/visp_ws/build
Devel space: /home/jifu/ros_workspaces/visp_ws/devel
Install space: /home/jifu/ros_workspaces/visp_ws/install
####
#### Running command: "cmake /home/jifu/ros_workspaces/visp_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/jifu/ros_workspaces/visp_ws/devel -DCMAKE_INSTALL_PREFIX=/home/jifu/ros_workspaces/visp_ws/install -G Unix Makefiles" in "/home/jifu/ros_workspaces/visp_ws/build"
####
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/jifu/ros_workspaces/visp_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jifu/ros_workspaces/visp_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 7 packages in topological order:
-- ~~  - vision_visp (metapackage)
-- ~~  - visp_bridge
-- ~~  - visp_camera_calibration
-- ~~  - visp_hand2eye_calibration
-- ~~  - visp_ros
-- ~~  - visp_tracker
-- ~~  - visp_auto_tracker
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'vision_visp'
-- ==> add_subdirectory(vision_visp/vision_visp)
-- +++ processing catkin package: 'visp_bridge'
-- ==> add_subdirectory(vision_visp/visp_bridge)
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: filesystem program_options 
-- +++ processing catkin package: 'visp_camera_calibration'
-- ==> add_subdirectory(vision_visp/visp_camera_calibration)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- visp_camera_calibration: 4 messages, 1 services
-- +++ processing catkin package: 'visp_hand2eye_calibration'
-- ==> add_subdirectory(vision_visp/visp_hand2eye_calibration)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- visp_hand2eye_calibration: 1 messages, 3 services
-- +++ processing catkin package: 'visp_ros'
-- ==> add_subdirectory(visp_ros)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
[... 351 lines cut ...]
  target, or an ALIAS target is missing?

CMake Error at vision_visp/visp_auto_tracker/CMakeLists.txt:99 (add_library):
  Target "auto_tracker" links to target "Boost::system" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at vision_visp/visp_auto_tracker/CMakeLists.txt:63 (add_library):
  Target "visp_auto_tracker_cmd_line" links to target "Boost::system" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at vision_visp/visp_auto_tracker/CMakeLists.txt:63 (add_library):
  Target "visp_auto_tracker_cmd_line" links to target "Boost::thread" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at vision_visp/visp_auto_tracker/CMakeLists.txt:63 (add_library):
  Target "visp_auto_tracker_cmd_line" links to target "Boost::system" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
Invoking "cmake" failed`

fspindle, 2022-10-07:
You should install `libboost-thread-dev` and `libboost-system-dev` binary packages

jifu23, 2022-10-07:
> You should install `libboost-thread-dev` and `libboost-system-dev` binary packages

I tried that, but it doesn't work. I can find all the binary in "/usr/lib/x86_64-linux-gnu", but it always has problems with Boost::system and Boost::thread.
[image]