lagadic/vision_visp #129: Build failed when VISP is installed from source code

jifu23, 2023-08-19:
Hi, I got a problem when I was building this repo.
Environment: ubuntu 22.04, ROS rolling.
When the VISP used in ros is installed from the latest source code, I get the following error when building vision_visp:

> $ colcon build --symlink-install
> Starting >>> visp_bridge
> Starting >>> visp_tracker
> Finished <<< visp_bridge [8.78s]                                     
> Starting >>> visp_camera_calibration
> Starting >>> visp_hand2eye_calibration
> --- stderr: visp_camera_calibration                                    
> CMake Error at CMakeLists.txt:45 (add_executable):
>   Target "visp_camera_calibration_calibrator" links to target "Boost::system"
>   but the target was not found.  Perhaps a find_package() call is missing for
>   an IMPORTED target, or an ALIAS target is missing?
> 
> 
> CMake Error at CMakeLists.txt:45 (add_executable):
>   Target "visp_camera_calibration_calibrator" links to target "Boost::thread"
>   but the target was not found.  Perhaps a find_package() call is missing for
>   an IMPORTED target, or an ALIAS target is missing?
> 
> 
> CMake Error at CMakeLists.txt:64 (add_executable):
>   Target "visp_camera_calibration_image_processing" links to target
>   "Boost::system" but the target was not found.  Perhaps a find_package()
>   call is missing for an IMPORTED target, or an ALIAS target is missing?
> 
> 
> CMake Error at CMakeLists.txt:64 (add_executable):
>   Target "visp_camera_calibration_image_processing" links to target
>   "Boost::thread" but the target was not found.  Perhaps a find_package()
>   call is missing for an IMPORTED target, or an ALIAS target is missing?
> 
> 
> CMake Error at CMakeLists.txt:84 (add_executable):
>   Target "visp_camera_calibration_camera" links to target "Boost::system" but
>   the target was not found.  Perhaps a find_package() call is missing for an
>   IMPORTED target, or an ALIAS target is missing?
> 
> 
> CMake Error at CMakeLists.txt:84 (add_executable):
>   Target "visp_camera_calibration_camera" links to target "Boost::thread" but
>   the target was not found.  Perhaps a find_package() call is missing for an
>   IMPORTED target, or an ALIAS target is missing?
> 
> 
> CMake Generate step failed.  Build files cannot be regenerated correctly.
> ---
> Failed   <<< visp_camera_calibration [2.04s, exited with code 1]
> Aborted  <<< visp_hand2eye_calibration [2.03s]
> Aborted  <<< visp_tracker [29.9s]                                 
> 
> Summary: 1 package finished [30.1s]
>   1 package failed: visp_camera_calibration
>   2 packages aborted: visp_hand2eye_calibration visp_tracker
>   3 packages had stderr output: visp_camera_calibration visp_hand2eye_calibration visp_tracker
>   2 packages not processed

But, when the VISP is installed via `sudo apt-get install ros-rolling-visp`, I didn't get these error. I would like to know why this is happening and how to fix it.
Thank you!

jifu23, 2023-08-19:
I solved it by explicitly linking against Boost work mentioned in [https://github.com/lagadic/visp/issues/780](url). But I got a new problem:

> --- stderr: visp_tracker                               
> Bagfile installed in /home/jifu/ros_ws/rolling/visp_ws/install/visp_tracker/share/bag/tutorial-static-box
> /bin/sh: 1: rosbags-convert: not found
> gmake[2]: *** [CMakeFiles/external_bag.dir/build.make:120: externals/src/external_bag-stamp/external_bag-patch] Error 127
> gmake[1]: *** [CMakeFiles/Makefile2:746: CMakeFiles/external_bag.dir/all] Error 2
> gmake[1]: *** Waiting for unfinished jobs....
> gmake: *** [Makefile:146: all] Error 2
> ---
> Failed   <<< visp_tracker [38.1s, exited with code 2]

ZxinZzz, 2023-10-10:
I have same question, did you resolve it?

fspindle, 2023-10-10:
Are you building the `rolling` branch ?

ZxinZzz, 2023-10-17:
> 你在建立分支吗？`rolling`

yes

yuxuanfanOrion, 2023-10-23:
> I solved it by explicitly linking against Boost work mentioned in [https://github.com/lagadic/visp/issues/780](url). But I got a new problem:

I met the same question, have you solved it?