lagadic/vision_visp #134: Request for Humble support.

TobiasAxelsen, 2024-04-02:
Hello!

I've been trying to install the library for the use of Ros Humble.

When doing step 3 in the tutorial I am met by the following error.

```
CMake Error at CMakeLists.txt:16 (find_package):
  By not providing "Findcamera_calibration_parsers.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "camera_calibration_parsers", but CMake did
  not find one.

  Could not find a package configuration file provided by
  "camera_calibration_parsers" with any of the following names:

    camera_calibration_parsersConfig.cmake
    camera_calibration_parsers-config.cmake

  Add the installation prefix of "camera_calibration_parsers" to
  CMAKE_PREFIX_PATH or set "camera_calibration_parsers_DIR" to a directory
  containing one of the above files.  If "camera_calibration_parsers"
  provides a separate development package or SDK, be sure it has been
  installed.```

I would appreciate any help!

m11112089, 2024-04-02:
This command installs the camera_calibration_parsers package. 
> $ sudo apt install ros-humble-camera-calibration-parsers

TobiasAxelsen, 2024-04-02:
Hello sir, thank you for the swift reply!

Is there a documentation somewhere, I am unable to find it!

Best regards

TobiasAxelsen, 2024-04-03:
Hello again @m11112089 

I have now run the command you prompted me to run, but I am still unable to symlink install, and I'm once again, given the following error.

```
--- stderr: visp_tracker                                               
CMake Error at CMakeLists.txt:17 (find_package):
  By not providing "Findimage_proc.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "image_proc", but CMake did not find one.

  Could not find a package configuration file provided by "image_proc" with
  any of the following names:

    image_procConfig.cmake
    image_proc-config.cmake

  Add the installation prefix of "image_proc" to CMAKE_PREFIX_PATH or set
  "image_proc_DIR" to a directory containing one of the above files.  If
  "image_proc" provides a separate development package or SDK, be sure it has
  been installed.

```

m11112089, 2024-04-03:
Could you please clarify whether you're looking for guidance on installing the vision_visp package and its dependencies, or if you're seeking instructions for using vision_visp after it's been installed?

If you're in need of a comprehensive installation guide that includes steps for installing third-party dependencies like camera_calibration_parsers, I'm here to assist.

Furthermore, documenting any issues encountered during the compilation of the vision_visp ROS2 rolling branch could greatly benefit both current and future users. To this end, we might consider initiating an issue to gather these errors or creating a Pull Request to update the README.md on the repository. This method has proven to be quite effective in addressing and resolving similar issues in the past. @fspindle, what are your thoughts on this approach?

exemplified by:
Issue [#129](https://github.com/lagadic/vision_visp/issues/129): Build failed when VISP is installed from source code.
Issue [#130](https://github.com/lagadic/vision_visp/issues/130): stderr: visp_tracker in rolling branch.
Request [#134](https://github.com/lagadic/vision_visp/issues/134): Request for Humble support.

m11112089, 2024-04-03:
> Hello again @m11112089
> 
> I have now run the command you prompted me to run, but I am still unable to symlink install, and I'm once again, given the following error.
> 
> ```
> --- stderr: visp_tracker                                               
> CMake Error at CMakeLists.txt:17 (find_package):
>   By not providing "Findimage_proc.cmake" in CMAKE_MODULE_PATH this project
>   has asked CMake to find a package configuration file provided by
>   "image_proc", but CMake did not find one.
> 
>   Could not find a package configuration file provided by "image_proc" with
>   any of the following names:
> 
>     image_procConfig.cmake
>     image_proc-config.cmake
> 
>   Add the installation prefix of "image_proc" to CMAKE_PREFIX_PATH or set
>   "image_proc_DIR" to a directory containing one of the above files.  If
>   "image_proc" provides a separate development package or SDK, be sure it has
>   been installed.
> ```
> 
> May this be related to the fact that when I run the
> 
> ```
> $ sudo apt-get install visp-doc
> ```
> 
> This happens when in 3rd of the installation process, which is denoted step 2?

It appears you've encountered another package dependency issue, this time with image_proc. The error you're seeing indicates that the image_proc package, which is required by visp_tracker, isn't installed or can't be found by CMake.

I've experienced a similar error in the past. Installing the image_proc package directly should resolve this problem. You can install it by running the following command:
> $ sudo apt install ros-humble-image-proc

TobiasAxelsen, 2024-04-03:
In general, I am trying to install VISP, such that I can use it in conjunction with ROS2 to do some virsual servoing in a robot simulation.

I am finding trouble in installing VISP in general :(

santidadday, 2024-04-10:
Hello sir, 

I also want to be able to use vision_visp in ros humble. However, we encountered many problems during the implementation process. Is there any good way to use it? Or when will the humble version of vision_visp be released?

Best regards

m11112089, 2024-04-15:
> Hello sir,
> 
> I also want to be able to use vision_visp in ros humble. However, we encountered many problems during the implementation process. Is there any good way to use it? Or when will the humble version of vision_visp be released?
> 
> Best regards

For detailed guidance on installation and troubleshooting, you may find the following discussion useful:

[Assistance with Installing vision_visp in ROS Humble #136](https://github.com/lagadic/vision_visp/issues/136).