lagadic/vision_visp #138: Error: Packages with unknown build types exist Can not build workspace with packages of unknown build_type

GoodandNo, 2024-04-21:
Does anyone know how to solve the problem?
when i use the command "catkin_make_isolated --pkg visp_hand2eye_calibration"
My system is ubuntu 18.04 and i have already installed ros-melodic v1.14.13
[image]

[image]

xushenghuaaa, 2024-05-10:
who knows?
 help

fspindle, 2024-05-11:
You need to build ViSP and set `VISP_DIR` env var. To this end follow [ViSP quick installation](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-ubuntu.html#install_ubuntu_quick) instructions.

Then source your `bashrc` and try a new `catkin_make`