lagadic/vision_visp #140: vision_visp : build visp_hand2eye_calibration_calibrator fails

viprob-ai, 2024-06-26:
[image]

any help ?

fspindle, 2024-06-26:
How do you build/install ViSP and which version ?

viprob-ai, 2024-06-26:
I have built the visp from source 
Version 3.6.0

aleciara, 2024-08-01:
Hello everyone,

I had a similar issue and resolved it by modifying the CMakeLists.txt files. For linking Boost libraries, I used the following approach:

`
find_package(Boost REQUIRED COMPONENTS 
  system 
  thread 
)

target_link_libraries(your_target_name
                      ${Boost_LIBRARIES})
`

I applied this method to the visp_camera_calibration, visp_hand2eye_calibration, visp_auto_tracker, and visp_ros packages.If this approach seems useful, I would be happy to assist.

Best,
Alessandro

fspindle, 2024-08-23:
Should be fixed. Reopen if needed.

commit f057eb9e (2024-08-23): Fix build around Boost::thread and Boost::system targets that are required but not found by cmake. Closes #140
closed 2024-08-23 by fspindle